DCC-EX/EX-T3-WiFi #1: platformio compile errors

jregholdsworth, 2022-11-21:
lots of compile errors in platformio as listed below. I know its just a few simple problems or just me being thick..but..Help!:

Compiling .pio\build\firebeetle2\src\WiFiUI.cpp.o
Compiling .pio\build\firebeetle2\src\main.cpp.o
src/Locos.cpp: In member function 'void Locos::add(uint16_t)':
src/Locos.cpp:21:13: error: 'find' is not a member of 'std'
   auto it = std::find(_locos.begin(), _locos.end(), address);
             ^
src/Locos.cpp: In member function 'uint16_t Locos::remove()':
src/Locos.cpp:36:13: error: 'find' is not a member of 'std'
   auto it = std::find(_locos.begin(), _locos.end(), address);
             ^
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiScan.cpp.o
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiServer.cpp.o
Compiling .pio\build\firebeetle2\lib7d3\WiFi\WiFiUdp.cpp.o
In file included from lib/UI/UI.h:8:0,
                 from src/SettingsUI.h:4,
                 from src/SettingsUI.cpp:1:
lib/UI/Elements/Button.h: In constructor 'Button::Appearance::Appearance(const String&, uint16_t, uint16_t, uint16_t, const String&)':lib/UI/Elements/Button.h:17:18: error: 'Button::Appearance::border' will be initialized after [-Werror=reorder]
         uint16_t border;
                  ^
lib/UI/Elements/Button.h:14:16: error:   'String Button::Appearance::icon' [-Werror=reorder]
         String icon;
                ^
lib/UI/Elements/Button.h:19:9: error:   when initialized here [-Werror=reorder]
         Appearance(const String &label, uint16_t color, uint16_t fill, uint16_t border, const String &icon = "")
         ^
In file included from lib/UI/UI.h:8:0,
                 from src/LocoByNameUI.h:4,
                 from src/LocoByNameUI.cpp:1:
lib/UI/Elements/Button.h: In constructor 'Button::Appearance::Appearance(const String&, uint16_t, uint16_t, uint16_t, const String&)':lib/UI/Elements/Button.h:17:18: error: 'Button::Appearance::border' will be initialized after [-Werror=reorder]
         uint16_t border;
                  ^
lib/UI/Elements/Button.h:14:16: error:   'String Button::Appearance::icon' [-Werror=reorder]
         String icon;
                ^
lib/UI/Elements/Button.h:19:9: error:   when initialized here [-Werror=reorder]
         Appearance(const String &label, uint16_t color, uint16_t fill, uint16_t border, const String &icon = "")
         ^
In file included from lib/UI/UI.h:8:0,
                 from src/MenuUI.h:4,
                 from src/MenuUI.cpp:1:
lib/UI/Elements/Button.h: In constructor 'Button::Appearance::Appearance(const String&, uint16_t, uint16_t, uint16_t, const String&)':lib/UI/Elements/Button.h:17:18: error: 'Button::Appearance::border' will be initialized after [-Werror=reorder]
         uint16_t border;
                  ^
lib/UI/Elements/Button.h:14:16: error:   'String Button::Appearance::icon' [-Werror=reorder]
         String icon;
                ^
lib/UI/Elements/Button.h:19:9: error:   when initialized here [-Werror=reorder]
         Appearance(const String &label, uint16_t color, uint16_t fill, uint16_t border, const String &icon = "")
         ^
In file included from lib/UI/UI.h:8:0,
                 from src/ProgramUI.h:4,
                 from src/ProgramUI.cpp:1:
lib/UI/Elements/Button.h: In constructor 'Button::Appearance::Appearance(const String&, uint16_t, uint16_t, uint16_t, const String&)':lib/UI/Elements/Button.h:17:18: error: 'Button::Appearance::border' will be initialized after [-Werror=reorder]
         uint16_t border;
                  ^
lib/UI/Elements/Button.h:14:16: error:   'String Button::Appearance::icon' [-Werror=reorder]
         String icon;
                ^
lib/UI/Elements/Button.h:19:9: error:   when initialized here [-Werror=reorder]
         Appearance(const String &label, uint16_t color, uint16_t fill, uint16_t border, const String &icon = "")
         ^
Compiling .pio\build\firebeetle2\lib55a\ESP Async WebServer\AsyncEventSource.cpp.o
*** [.pio\build\firebeetle2\src\Locos.cpp.o] Error 1
In file included from lib/UI/UI.h:8:0,
                 from src/LocoUI.h:5,
[... 211 lines cut ...]
lib/UI/UI.h: In instantiation of 'T* UI::addElement(Args&& ...) [with T = Label; Args = {int, int, int, int, const char (&)[19], bool}]':
src/WiFiUI.cpp:28:72:   required from here
lib/UI/UI.h:23:114: error: invalid use of 'void'
lib/UI/UI.h: In instantiation of 'T* UI::addElement(Args&& ...) [with T = Label; Args = {int, int, int, int, const char (&)[21], bool}]':
src/WiFiUI.cpp:29:70:   required from here
lib/UI/UI.h:23:114: error: invalid use of 'void'
lib/UI/UI.h: In instantiation of 'T* UI::addElement(Args&& ...) [with T = Label; Args = {int, int, int, int, const char (&)[40], bool}]':
src/WiFiUI.cpp:33:85:   required from here
lib/UI/UI.h:23:114: error: invalid use of 'void'
src/main.cpp: In function 'void setup()':
src/main.cpp:357:6: error: 'ARDUINO_EVENT_WIFI_STA_CONNECTED' was not declared in this scope
   }, ARDUINO_EVENT_WIFI_STA_CONNECTED);
      ^
src/main.cpp:363:6: error: 'ARDUINO_EVENT_WIFI_STA_DISCONNECTED' was not declared in this scope
   }, ARDUINO_EVENT_WIFI_STA_DISCONNECTED);
      ^
*** [.pio\build\firebeetle2\src\ThrottleServer.cpp.o] Error 1
cc1plus.exe: some warnings being treated as errors
*** [.pio\build\firebeetle2\src\WiFiUI.cpp.o] Error 1
cc1plus.exe: some warnings being treated as errors
*** [.pio\build\firebeetle2\src\main.cpp.o] Error 1
==================================================== [FAILED] Took 12.22 seconds ====================================================

 *  The terminal process "C:\Users\John\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

alex-code, 2022-11-21:
I think this is an espressif32 version issue.
Could you try replacing this in [platformio.ini](https://github.com/DCC-EX/EX-T3-WiFi/blob/master/platformio.ini#L15)
```ini
platform = espressif32
```
with
```ini
platform = espressif32@^5.2.0
```

jregholdsworth, 2022-11-21:
so that worked as far as I can tell..ie it compiled ..but I had  this message "Please disable/uninstall mitaki28.vscode-clang" and had to do it before it would compile..I've ordered the hardware so when it arrives i'll be able to confirm that its fixed or other wise.

alex-code, 2022-11-21:
Seems like some extensions are conflicting.
There's a mention here https://github.com/platformio/platformio-vscode-ide/issues/118 but my install is pretty vanilla and not had any errors like that.

closed 2023-01-29 by github-actions[bot]