DCC-EX/EX-T3-WiFi #3: Sta mode IP address for throttle

jregholdsworth, 2022-12-04:
In setting up for sta mode IE using my home network the use of DNS didn't work so dccex couldn't be found.

So it changed to sta mode .however what is it's on home network IP address ..I couldn't find it using advanced  op scanner?.. 

I edited the SD card to insert the real IP address of the C's and that worked.

However to manage locos I need to access the throttles web page...without it appearing on the scan I'm stuck.

Any advice?

jregholdsworth, 2022-12-04:
Sorry advanced ipscanner and not C's CS

alex-code, 2022-12-04:
Were you on the `CS Settings and WiFi display` (The one with the QR codes)? The web server on the throttle is only active on this display.

jregholdsworth, 2022-12-04:
Yep I thought the throttle web server would move to my home network when I had the CS and the throttle connected to it..which would be very convenient.

Also having difficulty saving stuff to the SD card..eg loco names. Looked on the serial port and the output said that access to the SD was denied when creating the locos directory on the sd

Also I have to enter the password twice when accessing the ap for the throttle etc page using my windows laptop

alex-code, 2022-12-04:
The throttle server should be accessible when connected to your WiFi. Did you include `http://`?
When the throttle is connected and you're on the WiFi display are you able to ping it?

Ok, didn't notice git ignores empty directories so wasn't created.
There should be `locos` and `fns` directories on the SD card too.

Not sure about the password issue. It's just a WiFi AP so if you've entered the settings Windows should just connect.

jregholdsworth, 2022-12-04:
Even though the t3-ex-wifi connects as an sta to my home 
network..192.168.0.xxx and can control my dccex C's which is also an sta on 
my home network..the throttle web server remains at 192.168.4.1

Can't save to the SD card reliably is another problem..

If the SD card loco structure is the same as your tethered throttle I can 
create a simple config by hand on the sd and give that a try.

I am really impressed despite these teething issues which in part are due 
to my own dull nature

Regards JH

alex-code, 2022-12-04:
I'm working on a fix for the SD missing directory issue, just didn't notice git omits empty directories which means some that are required will be missing if you copy the `sd` directory.

alex-code, 2022-12-04:
I've updated the code with a fix for the missing directories.
The throttle will now create them if needed.

Have you been able to access the throttle's website?
If in STA mode the throttle's IP will come from your router, it doesn't set its own.