DCC-EX/EX-T3-WiFi #6: compilation problem in new release

jregholdsworth, 2022-12-05:
Fails during compilation with the following output:

lots of good stuff then

Compiling .pio\build\firebeetle2\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\firebeetle2\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\firebeetle2\FrameworkArduino\wiring_shift.c.o
In file included from lib/UI/Img/Img.cpp:2:
lib/UI/Img/BMP.h:9:31: error: expected class-name before '{' token
 class BMP : public ImgHandler {
                               ^
lib/UI/Img/Img.cpp: In static member function 'static Img* Img::load(fs::File)':
lib/UI/Img/Img.cpp:9:25: error: cannot convert 'BMP*' to 'Img*' in return
       return new BMP(img);
                         ^
*** [.pio\build\firebeetle2\lib258\UI\Img\Img.cpp.o] Error 1

alex-code, 2022-12-05:
`Img.h` & `Img.cpp` were renamed.

How did you update?

jregholdsworth, 2022-12-05:
I down loaded the new code files from git hub to the old location..reloaded 
the project files and then tried a test compile..I'll create a completely 
new folder and try again

jregholdsworth, 2022-12-05:
Apologies I cleaned and made a new folder and it works...the throttle has 
an ip address of 192.168.0.202...god only knows why the router chose this 
but it works..DHCP works in mysterious ways?

Thanks again..

Ps I note that although the throttle web page is inactive when the T3 
button isn't pressed I can still ping it's address and get a response..

alex-code, 2022-12-05:
Glad you got it working.
A device can respond to a ping without having an active server.

closed 2022-12-05 by alex-code