DCC-EX/EX-T3-WiFi #11: error: expected class-name before '{' token  class BMP : public ImgHandler {

daveclifford, 2022-12-10:
**Describe the bug**
Running Visual Studio Code app in Linux
Note BMP.cpp and BMP.h are dated 2022/12/08; Img.cpp and Img.h are dated 2022/11/29

snippet of the code complie log  pasted below:
"In file included from lib/UI/Img/Img.cpp:2:
lib/UI/Img/BMP.h:9:31: error: expected class-name before '{' token
 class BMP : public ImgHandler {
                               ^
lib/UI/Img/Img.cpp: In static member function 'static Img* Img::load(fs::File)':
lib/UI/Img/Img.cpp:9:25: error: cannot convert 'BMP*' to 'Img*' in return
       return new BMP(img);
                         ^
*** [.pio/build/firebeetle2/lib843/UI/Img/Img.cpp.o] Error 1"

**To Reproduce**
Steps to reproduce the behavior:
1. re-run build job

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'

**Expected behavior**
Compiles to end w/o error code

**Screenshots or debug output**
If applicable, add screenshots or debug output to help explain your problem.
*Debug output can be obtained from the serial monitor*

**Additional info**
 Executing task in folder EX-T3-WiFi-master: platformio run --environment firebeetle2 
Processing firebeetle2 (platform: espressif32@^5.2.0; board: firebeetle2; framework: arduino)

alex-code, 2022-12-10:
`Img.h` and `Img.cpp` were renamed.
You'll need to delete those or start with a new directory with downloading/cloning.

daveclifford, 2022-12-14:
Thanks - did "You'll need to delete those or start with a new directory with downloading/cloning" and fixed issue.