DCC-EX/EX-T3-WiFi #31: README.md - RE: "This battery has the terminals the correct way but before connecting any battery check the wires are the right way round on the plug"

daveclifford, 2023-01-14:
[For your review - This may clarify what the user should do to ensure the positive voltage is in the correct position.]  

The JST-PHR-2P Li-Po battery connector has a red-colored lead representing the positive polarity and when plugged into the socket on the side of the FireBeetle2, should face the back (toward the ESP32 chip) There is a Plus (+) indicator on the underside of the board indiccating the battery positive connection.

The user may use a Volt-Ohm-Meter (VOM) to measure the voltage (if charged) to ensure the positive voltage is at the red wire and in the right position (when looking at the connector head-on).

alex-code, 2023-01-14:
Good idea, will look at adding some extra pictures of the connectors and more instructions to help check the battery polarity is correct.

daveclifford, 2023-03-16:
In order to keep this open,

I will offer you some additional information on the battery and system power draw.

I purchased a USB Tester when I was having connectivity issues and wanted to eliminate poer/interface issues.

The unit is a DRok USB tester and has a USB dongle with USB A and USBC plugs as input to the tester.
There is a voltage line (in red) and a power line (in blue).

The FireBeetle2  draws:

5.07 Volts and 0.15-0.16 Amps  (150-160ma)

The chip is slightly warm to my index finger.

Dave