DCC-EX/EX-T3-WiFi #34: Ideas : Documentation Improvements

Alan-Lomax, 2023-01-15:
Not a problem but an observation:

- In reviewing the assembly.md it is clear that a lot is left unsaid. Scrolling down to the first picture I see an accelerometer screw down and some details about wiring it. It occurs to me that is not an assembly step but a wiring detail.
- I think this document should start with 'a box of parts' and lead step by step putting it all together. 
- The details of how to tackle specific items in the process is good and must not be lost .. but it is not part of the main 'assembly flow'
- Other examples observed further down - for example - firebeetle is attached to frame - ok but in the photo the frame is already attached to something - when did that happen?  Are other parts attached to the frame first?  

NEXT
- Usage Cases - I saw several places where the accelerometer is described as optional. I could not find why I might want to use it.
- A block diagram of the subassemblies and how they integrate could provide some context. Wiring the switch and what it is connected to etc.  

In my case I have ordered my parts and started 3D printing the case. 
While awaiting delivery I have been reading and re-reading what I can. 
I like what I have seen but some polish here could really raise the profile

alex-code, 2023-01-15:
There's definitely room for improvement.
The pictures are a bit out of step as I built it first before going back to get some pictures so some parts that would be done later already exist.

When I can I need to do a full disassembly with a step by step reassembly.

That's a good point about the accelerometer, I should mention it allows the display to rotate with the throttle so the dial can be at the top or the bottom.

Added to the list 👍

daveclifford, 2023-01-18:
I made this file to ensure I had the correct size and number of screws and the correct assembly order.

One issue you **should highlight** is the need to file the Touchscreen opening for correct size in the Top piece. My display was $40 USD and the most expensive part and could easily be damaged by someone less careful than me..
 
Please review and see if this may help others. I would be willing to help you - we could chat via WhatsApp. 
I am  7 hrs  (Aurora CO) U.S. Mountain Standard  Time before London (GMT) time.
 
[Screw List and Assembly Order.txt](https://github.com/DCC-EX/EX-T3-WiFi/files/10451332/Screw.List.and.Assembly.Order.txt)

alex-code, 2023-01-18:
> I made this file to ensure I had the correct size and number of screws and the correct assembly order.
>
> One issue you **should highlight** is the need to file the Touchscreen opening for correct [...]

Thanks for mentioning about the display, I guess the tolerances could vary between 3D printers so some careful trial fitting should be done.

If I can this weekend I'll disassemble then re-assemble getting more pictures along the way.

daveclifford, 2023-01-18:
I'm guessing that the audience for this throttle is wide - with varying degrees of  experience. I can follow your photos ( great job, by the way) and  .md files. I think the area that  needs more explanation is the setup for VSC and PIO to complie and upload the code.  This is NOT simple to understand and use. I know you warned the audience, BUT there are some with little code development experiece that only want to upload and use. Perhaps a compiled code file and easy upload procedure  could facilitate the builder.  The skills needed to build this WiFi Throttle are:
soldering expertease, assembly expertease with very small 2mm screws needing magnification, understanding step building, installing and using VSC and PIO (I think this may be the biggest challenge), and understanding the use of the throttle with the DCC-EX Arduino and (Motor and WiFi) Shields. 
I have the skills to build, assemble and use the WiFi Throttle - but I worked for IBM for 40 years as a Software Developer and Team Lead and have tinkered with many software and hardware systems.

I think a list of tools needed to build the WiFi throttle may be helpful. 

I am retired and have the time to give you to improve this project. 
Unfortunly, we may never meet in person.

Dave

alex-code, 2023-01-21:
Made a start on updating the docs with new pictures and numbered steps.

> I'm guessing that the audience for this throttle is wide - with varying degrees of experience. I can follow your photos ( great job, by the way) and .md files. I think the area that needs more [...]

I had thought about a compiled build but that would require some changes.
With a compiled build I'd have to move the local WiFi generation code to the throttles first turn on, currently it's done with `Python` on the build step in PIO.
I'm not sure if a compiled build would be much easier as you'll still need software to upload.

I can add some extra instructions on using VSCode, PIO and downloading the code.

daveclifford, 2023-01-25:
My guess is that some (like I did) will stumble into this through the DCC-EX Command Station which uses the Arduino IDE - so couldn't that IDE be used to push the compiled build to the FireBeetle2?
 
I understand your use of Python to build "configuration" code. I just think you'd have a wider group building/using the WiFi Throttle if they didn't have to install/learn/manage two new tools (VSC and PIO).

Why don't you use the Discussion/Poll feature to see if there is interest in a compiled build? Maybe I'm dreaming and your current audience is so small NOT for you to warrant the code changes to go to a compiled build.

I am committed to seeing this through .....  Using Linux as the development platform caused security issues with using a USB port with the PIO app, so I bought a small MiniPC (Beelink Mini PC, Mini S Intel 11th Gen 4-Cores N5095, Mini Desktop Computer with W11Pro) just to build/install the code. I am going to set it up by 27th January. --- stay tuned!!

I saw your videos and was intrigued.............

(I sometimes write this (as I am now at 2125 MST US) while sipping a red beverage - it may not be coempletely coherent).

[There appears no automagic spell checker - RATS!]

Kind Regards,

Dave