DCC-EX/EX-T3-WiFi #37: Docs - Names of files are one word but titles in files are more descriptive

daveclifford, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
I have to read the file to understand what it does - I think the file name should describe what the file contents is describing.

**Describe the feature you'd like**

Original File Name  - New file name

Assembly - WiFi_Throttle_Assembly
Code -  Uploading_The_WiFi_Throttle_Code
Locos - Setting_Up_And_Managing_Locos
Menu - WiFI_Throttle_Menu
Decoder Programming  -  Programming_The_Loco_Decoder
Running - Running_Locos
Settings - WiFi_Throttle_Settings
Setup - WiFi_Throttle_Setup

If you are worried about spaces in the file names , use underscores for spaces (as I suggest).

I also strongly feel that you should use the word "WiFi"  in your documentation to emphasize that this is an untethered wireless throttle.

**I am in awe** when I touch the throttle case components; you **did an excellent job** in planning and creating the case components - particularly the frame to hold the FireBeetle2, the accelerometer, the rotary encoder and the switch. **Kudos to you!**

alex-code, 2023-01-21:
Good idea. I guess not every user will be using the `README` to get to the other instructions.

Thanks for the compliments, It took a fair few hours of thinking and playing around before I came up with the design.
I would like to make it slimmer but using off the shelf components I have to be a bit more liberal with space.

daveclifford, 2023-01-22:
I will send you a JPG of a case for another WiFi throttle that leaves a lot to desire - after you see it, you will be very satisfied to know that you did a great job  in making a case that fits well in a small hand (mine) . I'm glad you chose a large TouchScreen (for us older folks) and that probably dictated the case dimensions (width and length).