DCC-EX/EX-T3-WiFi #38: Heat and the LED on the FireBeetle2

daveclifford, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
The case encloses electronic components that generate heat and there are no holes/slots for the heat to escape.

**Describe the feature you'd like**

1. Perhaps a small slot on the side opposited the switch and USB-C socket would allow heat to escape. 
    FireBeelte2 - ESP32-E - Max Discharge Current: 600mA@3.3V LDO
    FireBeetle2 -  ESP32-E -Max Charge Current: 500mA

3. Is the LED on the FireBeetle 2 used to signal errors in uploading the code2. 
If so, maybe there could be a way for a small hole to be drilled, so the user could see the light or maybe the screen could provide a warning  that the reset button wouldd have to be pushed. Maybe the battery could be moved down so a rod couild be poked into a hole in the base to momentarily reset the switch. 

**Additional context**
Add any other context or screenshots about the feature request here.

daveclifford, 2023-01-20:
In placing the FireBeetle2 into the frame, I discovered that the two buttons (RST and D4/27) on the FireBeetle2 are obscured by: 

1. a post to mount the on/off switch, and,
2. a post to tighten the bottom of the case to the frame. 

A solution is to allow a small thin tool to be used to push the RST button and the user would have to file a very little material from the inside of the corner post to allow clearance for the tool to be able to push in the RST button.

I do not know if the RST button would ever be needed to be used, but it wouldn't hurt to make the modification to the frame **before** it was needed. If needed and the frame is unmodified, the user would have to disassemble the FrieBeetle2 from the frame to reset the microprocessor. Painful and risky!

[image]

alex-code, 2023-01-21:
To be honest I've not tested for heat so can't say whether it'll be an issue.
The only issue with holes on a vertical wall is that they can be tricky to 3D print uniformly.

The one LED that's probably relevant is the charging one as it has some flash sequences.

The `RST` button shouldn't be needed (think it'll only reset the ESP32 too) as the wired in switch effectively does the same thing but will also reset the display.

daveclifford, 2023-01-22:
Today, I started to assemble the components to the throttle frame. In particular, I was able to use 4 M2x6mm self-tapping Cap Head Screws (vs M2x5mm as the assembly steps suggest). The bottom of the frame is sufficiently thick enough to provide clearance where the sharp end of the screw does not show  outside the frame (whew!).

I also think the some small holes could be drilled into the bottom of the case to provide a) ventilation, and b) the ability to see the LEDs. This is not needed now, but could be an enhancement for a later date. There is a thickl web going across the short end of the case that could be drilled and still provide substantial support.

daveclifford, 2023-01-22:
I forgot to mention - you may wish to add a note in  the Assembly.MD that the case components may need to be filed - in particular, the holes for the On/Off Switch and FireBeetle2 USB-C Connector. I noticed that there is some schmutz left from the 3D printing process; the two holes for the On/Off Switch and USB-C Connector and the four holes in the bottom of the case - the recess for the screw heads were not clean.

daveclifford, 2023-02-01:
When I get the WiFi Throttle asssembled, I'll throw a Milliampmeter on the unit and measure the current draw. I'm curious to see how long the 3000 mah battery will last.

daveclifford, 2023-05-20:
i decided to measure the temp and report back to you

daveclifford, 2023-07-28:
Alex,

I think in the next 3-4 days I will insert a digital thermometer in the
case, turn it on and seal the gaps on the side with tape and let it run for
24 hrs, then take a measurement and report to you.

Dave

closed 2023-10-05 by github-actions[bot]