DCC-EX/EX-T3-WiFi #49: Browser based code upload

alex-code, 2023-02-04:
Investigate https://github.com/esphome/esp-web-tools for possible uploading of firmware using a browser.

Notes:
- Would require the throttle WiFi SSID and pass to be generated on the throttle.
- Serial support currently only in Chromium based browsers.

daveclifford, 2023-02-04:
Alex,

Great!  Thanks for finding this - *esp-web-tools*.

I believe that the  FireBeetle2 ESP32 chip isn't being set/reset into
bootloader mode, so the load won't work.

*[ref: https://docs.espressif.com/projects/esptool/en/latest/esp32s2/advanced-topics/boot-mode-selection.html
<https://docs.espressif.com/projects/esptool/en/latest/esp32s2/advanced-topics/boot-mode-selection.html>]*

"On many development boards with built-in USB/Serial, esptool.py can
automatically reset the board into bootloader mode. For other
configurations or custom hardware, you will need to check the orientation
of some “strapping pins” to get the correct boot mode:"

I'd prefer to use *esptools.py* as a PlatformIO pre-action prior to
uploading the build.  [The PlatformIO Build System has a rich API that
allows one to attach different pre-/post actions (hooks),,,,,] to use, *but
I haven' seen an example I understand and modify that I could use.*

I'll take a look at the code to see if I can glean some info.

I am at the point where all WiFi throttle components are soldered
(connectors) and attached to the frame. Now the fun part - wiring the
throttle components.
As you suggested, I got some small brass tubing to act as ferrules.

I am also building a DCC-EX Arduino w/MotorShield and WiFi boards, to
provide the power and signal.

I  like working with my hands and get immense satisfaction from building
with parts........

Thanks again,

Dave

daveclifford, 2023-02-11:
**"I'm looking into browser based uploading so you'd only need a Chromium based browser.
Will avoid the need to install IDE's and build the code yourself"**

I believe that you will become more famous than you are now.....

I have been developing software since 1970 and I don't think anyone who hasn't used an IDE would be able to go through the steps to successfully:

a) install two software tools, and
b) use PlatformIO to build/upload the build to a FireBeetle2.

More know how to use the Arduino IDE than the VSCode/PlatformIO combo.

Very few will help you add function to your code..., but most will l**ove to upload** the current build to their WiFi Throttle that they built themselves (pride of build and ownership).

**I whole-hardily applaud your interest in providing a binary build file that will be uploaded by a Chromium browser.**

Dave

alex-code, 2023-02-11:
Adafruit have a site where you can upload the binary to flash, I'm just working on a python script to produce a merged build.

https://adafruit.github.io/Adafruit_WebSerial_ESPTool/

daveclifford, 2023-02-11:
I dd a little snooping and saw:

the need for chrome://flags/  and enable the "Experimental Web Platform"

My issue was what serial port to enable (there were many)

We can sort that out..........................

I like this tool - much easier to set up and use - but the poor firefox
users c'est dommage.

Dave

alex-code, 2024-08-10:
Binary release now available.

closed 2024-08-10 by alex-code