DCC-EX/EX-T3-WiFi #50: Upload instructions

giraffe167, 2023-02-04:


alex-code, 2023-02-08:
I'm not the best at explaining/instructions.
There's a quick start guide here, https://docs.platformio.org/en/latest/integration/ide/vscode.html#quick-start

I'm looking into browser based uploading so you'd only need a Chromium based browser.
Will avoid the need to install IDE's and build the code yourself.

daveclifford, 2023-02-11:
**"I'm looking into browser based uploading so you'd only need a Chromium based browser.
Will avoid the need to install IDE's and build the code yourself"**

I believe that you will become more famous than you are now.....

I have been developing software  since 1970 and I don't think anyone who hasn't used an IDE would be able to go through the steps to **successfully**: 
 
a) install two software tools, and 
b) use PlatformIO to build/upload the build to a FireBeetle2.

More know how to use the Arduino IDE than  the VSCode/PlatformIO combo.

Very few will help you add function to your code..., but most will love to upload the current build to their WiFi Throttle that they built themselves  (pride of build and ownership).

I whole-hardily applaud your interest in providing a binary build file that will be uploaded by a Chromium browser.

Dave

giraffe167, 2023-02-18:
Thank you that worked

closed 2023-02-18 by giraffe167