DCC-EX/EX-T3-WiFi #63: No loco functions except forward and reverse.

Budman1758, 2023-02-20:
**Describe the bug**
When in the loco screen all I have is fwd & rev and speed. Seems like there should be more "stuff" Is this normal? The fwd ,rev  and speed functions work as they should as far as I can tell. Followed the code upload directions to the letter. (pretty sure anyway)
See attached picture.

[image]

**Screenshots or debug output**
If applicable, add screenshots or debug output to help explain your problem.

alex-code, 2023-02-20:
Can you tell me what version your CS is?
You'll need to be on the latest ideally as it has new functionality the throttle requires.

Budman1758, 2023-02-20:
I used the Windows installer so I assume it is the latest release version.

alex-code, 2023-02-21:
Do you have a MicroSD card in the throttle with the contents of the sd folder copied?
https://github.com/DCC-EX/EX-T3-WiFi/blob/master/docs/assembly.md#before-starting-assembly

Budman1758, 2023-02-21:
OK. That helps. Don't have any idea how I missed that part.... Musta had a senior moment there...lol. Now I have all the buttons on the loco screen. 

Are the icons only assigned to function buttons when customized individually? No "default" icon setup?

alex-code, 2023-02-21:
I didn't include icons with the default function buttons as they may vary between different decoders.

You can create a reusable function set if you've got the same decoders so you don't have to create one with each loco.

Budman1758, 2023-02-21:
Got a good source for appropriate icons? Lots of .svg and .png stuff but not .bmp.

alex-code, 2023-02-21:
I got some icons from here https://www.pngrepo.com then just resized (30x30) and converted to 24bit bmp.
This is a good free image editor program https://www.getpaint.net

closed 2023-02-22 by Budman1758