DCC-EX/EX-T3-WiFi #67: Disable swipe option

giraffe167, 2023-02-25:
Sometimes I have noticed that I accidently swipe on the screen and it can get a little annoying. 

Could you possibly make a button in the settings so you could disable swipe?

alex-code, 2023-02-26:
Good idea, I could add an extra `No Action` option to the swipe config.

alex-code, 2023-03-04:
Added `None` as an option for swipe actions.
This is a breaking change as I reordered the const numbers so you'll just need to touch each action option button once if you want to keep any settings you previously had.

giraffe167, 2023-03-04:
Hi i have uploaded the new code but the none button isn't showing as an option, but i have got:

Release/Keypad/Loco Names/Loco groups/Next Loco/ Prev Loco

alex-code, 2023-03-04:
Does it show when you cycle through the swipe options by touching the buttons?

giraffe167, 2023-03-04:
When pressing the swipe options, the none button doesn't show on the list

alex-code, 2023-03-04:
How did you update?
Did you download the new changes then upload?

giraffe167, 2023-03-04:
Sorted it i accidentally pressed the wrong upload button

Thank you for your help

closed 2023-03-04 by giraffe167