DCC-EX/EX-T3-WiFi #80: Setup instructions

Alan-Lomax, 2023-03-25:
In the setup.md instructions at the bottom it indicates 

> If your CS is connected to your local WiFi
> The server and port are dccex and 2560 by default, you'll only need to change these if you changed them on your Command Station

This was not my experience. I needed to replace the server name with the ip address of my CS on my local network.

alex-code, 2023-03-25:
Did you set a static IP with your CS when setting it up?
Could you try a `ping dccex` to see if it's discoverable on your network via the hostname.

Alan-Lomax, 2023-03-25:
Not exactly a static IP address - I set up an IP reservation (which amounts to the same thing). The same MAC address gets the same IP address (via DHCP) every time. Similar to a static IP address but far easier to maintain across a network.

Alan-Lomax, 2023-03-25:
I will try the ping test and report back. (how would that work with more than 1 CS ? )

Alan-Lomax, 2023-03-25:
OK .. as expected pinging the IP address **does work.** but pinging the server name "dccex" **does not**. 
It is not clear to me how it could ever work unless a manual entry was added to the hosts file.  (thus manually mapping dccex to a specific  IP address).

alex-code, 2023-03-26:
A bit of googling does mention a possible bug with the ESP8266 firmware and certain routers where it doesn't use the hostname the device provided.

For curiousity would you be able to try without the reservation to see if your router uses the hostname?

Alan-Lomax, 2023-03-27:
`For curiousity would you be able to try without the reservation to see if your router uses the hostname?`

**Done.**
Bottom line - no change

**In detail**

1) Checked my router - "Attached Devices" - CS shows up as ESP_E68F75 and has the reserved address 192.168.27.3
2) From a workstation on that network - and before changing anything - ping by IP works, ping to dccex does not.  ping to ESP_E68F75 does not.
3) On router removed DHCP reservation
4) In arduino IDE modified config.h for ControlStation (DCCEX+Wifi) and enabled a fixed IP address of 192.168.27.3
5) Recompiled and reloaded
6) From Workstation
       ping by IP works, ping to dccex does not.  ping to ESP_E68F75 also does not.
7) Checked my router again - "Attached Devices" - CS shows up as ESP_E68F75 - this info must have come from CS
8) Reverted config.h for DHCP
9) Recompiled and reloaded
10) Redid ping checks - everything as it was.

At no point was there even a sniff of something called dccex ... all the pings reported 'unable to resolve' 

As I mentioned before it is not obvious to me how this would work since most home networks do not have DNS server (Which is normally a domain controller) On your home router you enter the IP address of what you want to be your dns server and it is usually provided by your ISP. An individual (Windows or Linux) machine has a DNS resolver cache but that is not the same thing.
Our two IoT devices (Throttle and CS) don't have this capacity and neither does the router. My windows machine off on the side of the IoT devices has a cache (from previous dns lookups) but would not be aware of dccex.  I know I could 'make it work' by adding a manual entry in my local hosts file (essentially a dns override)

closed 2023-06-02 by github-actions[bot]