DCC-EX/EX-T3-WiFi #87: Throttle will not reconnect after turning off then back on.

Budman1758, 2023-04-10:
**Describe the bug**

I don't know if this is a DCC-EX "thing" or a problem with the throttle itself. When I turn on the DCC-EX power supply and then turn one of my throttles on everything works as it should. I can turn the main and program tracks on and off and all other functions work fine. I then turn off the throttle and set it aside for a while leaving DCC-EX on. When I come back and turn the throttle back on it does not connect to the power supply. The icon showing connection to the PS shows connected but no functions work. After a short time the connection changes to showing disconnected. 

BOTH of the throttles I currently have working show this behavior. Only way to get throttles to reconnect to the power supply is to power cycle it. As stated, this may actually be a DCC-EX bug. Not sure.....

**To Reproduce**
Steps to reproduce the behavior:
1. Turn DCC-EX on.
2. Turn on the throttle and it connects and functions. Turn either main or program tracks on.  (or both)
3. Turn throttle off leaving DCC-EX in track on state.
4. Come back after a bit, turn throttle back on.
5. Throttle does not connect to or control DCC-EX.

**Expected behavior**
Throttle should connect and function.

This is repeatable and constant.

**Additional info**
 - Component: Throttles built as per this website instructions. DCC-EX built as per the instructions there. Recommended Wi-Fi shield and latest version of the DCC-EX software.

alex-code, 2023-04-10:
Could you clarify what you mean by power supply? Is this your Command Station?

Is your CS connected to your home WiFi in station mode or is your CS in Access Point mode?

Budman1758, 2023-04-10:
DCC-EX command station.
Connected to home WiFi

alex-code, 2023-04-10:
How long do you have the throttle off for?
Doing a quick test on my setup the throttle re-connects (only off for about 10 seconds).

Budman1758, 2023-04-10:
> How long do you have the throttle off for? Doing a quick test on my setup the throttle re-connects (only off for about 10 seconds).

If I reconnect that soon it will sometimes work but usually not. But it never keeps working. It may work to turn off or on (track power) once but not anymore after that.

 If you wait over a minute it does not connect at all.

Ash-4, 2023-04-10:
Do you have this problem when using Engine Driver?
Has the DCC-EX WiFi ESP8266 been flashed to version 1.7.4 ? 
Is the Mega powered using 7-9 volts to the barrel connector?
Attach the startup logs and diags from the Arduino serial monitor.
For additional diags, use the commands
 <D WIFI 1>  and/or  <D WIT 1>

Budman1758, 2023-04-10:
> Do you have this problem when using Engine Driver? 

Yes. Logs on engine driver is showing "connection refused"

>Has the DCC-EX WiFi ESP8266 been flashed to version 1.7.4 ?

Almost sure now this is a DCC-EX command station issue

daveclifford, 2023-04-11:
"Has the DCC-EX WiFi ESP8266 been flashed to version 1.7.4?" 
How do I know what version I'm using ??

Dave

Ash-4, 2023-04-11:
Look for a few lines in the DCC-EX startup log.  This is what I see.
15:45:33.238 -> <* Wifi Check: [\r\nOK\r\n\0] *>
15:45:33.238 -> AT+GMR
15:45:33.238 -> AT version:1.7.4.0(May 11 2020 19:13:04)
15:45:33.238 -> SDK version:3.0.4(9532ceb)
15:45:33.238 -> compile time:May 27 2020 10:12:17
15:45:33.238 -> Bin version(Wroom 02):1.7.4
15:45:33.238 -> OK
15:45:33.238 -> <* Found in 17ms *>

Some of the WiFi shields can work with a different version, but when folks have problems with wifi, the first check is the version and the next check is for adequate power.  Without wifi, the usb connection should provide adequate power.  With wifi, there is some question.  I noticed a significant improvement in performance when Mega was powered with 9V to the barrel connector.  Greater than 7V is good.  12V will cause excess heat.  WiFi power requirements may be greater than expected, and several points of potential weakness with USB powered from PC USB port.

You may want to continue the discussion on the discord server; there have been some recent discussion on how to flash the wifi shield.
https://discord.com/invite/PuPnNMp8Qf

Ash-4, 2023-04-11:
Found a recent guide.
https://gist.github.com/nathankellenicki/7008540322c617869cec17226cff579d

There are other wiring options if you do not have the USB/TTL adapter.

daveclifford, 2023-04-11:
Thanks- I'll check out the stream for the version
Dave

alex-code, 2023-04-12:
Good to hear that fixed it.
I'm using a Mega+WiFi combo board and not experienced your issue with that.

Unfortunately the CS doesn't have a way for the throttle to get the power states without a little bit of work.
You need to include a file I've made when compiling and uploading your CS.
https://github.com/DCC-EX/EX-T3-WiFi/blob/master/docs/code.md#command-station

Budman1758, 2023-04-12:
> You need to include a file I've made when compiling and uploading your CS.

Well, that was easy.....  :)

Works like a champ.

closed 2023-04-12 by alex-code