DCC-EX/EX-T3-WiFi #92: Cannot seem to upload the filesystem

AndrewJPye, 2023-05-30:
Hi, not sure if something is missing or just my lack of knowledge! I tried to upload the filesystem but I don’t have the option in platformio. I tried to read up on it and I see it needs the files in the data folder. I see that this exists and the subfolders but I don’t see any css or other file in the data root folder. Should there be one for platformio to detect?? Any help gratefully received! Thanks in advance

alex-code, 2023-05-31:
Did you download the zip from github or do a clone?
Each of the subfolders should have files in.

I'd try downloading again then you should get the option.
https://github.com/DCC-EX/EX-T3-WiFi/blob/master/docs/code.md#uploading-the-code

AndrewJPye, 2023-05-31:
Thanks Alex. I installed gut in vscode and reloaded and it gives me the options now. I guess my config wasn't loading/up to date.... So, here goes attempt one for trying to get it running!!!

closed 2023-05-31 by AndrewJPye