DCC-EX/EX-T3-WiFi #107: Throttle UI button glitch

indexsites, 2024-04-23:
Using a 32 function loco decoder, button icon for F30 turns from pressed to idle state when some other function buttons are pressed or the throttle speed/direction is changed 

Steps to reproduce the behavior:
1. Create a button for a function > F28 (note, auto increment/decrement widget stops at 28 but value can be overwritten manually)
2. Control loco and press button for F30
3. Function F30 activates on loco and icon changes from idle to pressed state as expected
4. Change speed/direction or press another sound-related function button and the icon for F30 reverts to idle although the F30 function still runs (in my case F30 mutes the sound)
5. Display is now out of sync with loco state - display shows function is idle (off) but on loco function is still operating.  
7. Pressing the F30 button again sets the icon to the pressed state but as soon as the throttle is touched or other sound-related function are used the F30 button icon reverts to idle state again.

Expected behaviour:
The button icon for F30 always remains in pressed state until button for F30 is pressed again and is unaffected by throttle changes or selection/deselection of other sound related functions.

(Note, I haven't tested this with other functions > F28 but I would imagine the same behaviour results.)

Attached annotated 
[image]
[image]
screenshots show button icon for F30 (4th button on the 4th row) changes from pressed state to idle state after increasing speed of loco

Additional info:
 - Components all standard: DFRobot FireBeetle 2 - DFR0654-F, DFRobot Fermion: 3.5” 480x320 TFT LCD Capacitive Touchscreen - DFR0669, DFRobot Fermion: EC11 Rotary Encoder - SEN0235 (not using an accelerometer)
 - Version: 4.1.3

alex-code, 2024-04-24:
The throttle doesn't keep track of function states itself and relies on the broadcast from the CS which only tracks fns 0-28 so there's not much I can do.

The CS code is done so it'll work on an Uno which I don't know if it has enough memory to track up to 64 fns per loco.

Are you able to remap F30 to a Fn between 0-28?

indexsites, 2024-04-25:
Thank you for the quick response.  Remapping the function is a good workaround, thank you - I will do that.  

In the meantime thank you for all the time and effort you put into this project - I'm really pleased with my throttle and apart from a few minor problems (of my own making) the build was very straightforward.  

I'm especially appreciating the ability to configure the UI to my needs - uploading and arranging the button icons is fantastic.

alex-code, 2024-04-25:
Thanks for the positive feedback, glad it's all working for you 👍

closed 2024-04-25 by alex-code